Sensirion/arduino-avr-legacy-i2c-sps30 #14: after long time sps30 return only value 0.00

michapr, 2019-06-27:
After working a long time SPS30 returns now 0.00 (after restart sometimes very small values for PM X <0.2 - then all is 0.00 but particle size 1.63um all the time.

was starting 2 days ago with values 3712.84ug for PM10, after reset was ok for 10 hours, then values were 63652.7ug, then 67717.9ug then longer time (at the night) 4727.6ug and after about 45minutes 0.00ug ...
No more reset was helping...

Cleaning command is working (I can hear it) and the fan is running...

Is the device defective now?
Can I test anything else?

Michael

winkj, 2019-06-28:
Hi Michael,

Can you share what setup you're using, i.e. which arduino platform this is connected?

michapr, 2019-06-28:
I'm using ESP8266 with modified I2C lib (for getting all values) - it was working for several months.
Also have tried your latest sample code (0.0.6) as reference - same results.
Today I have tested to "clean" with fresh air the sensor from outside (careful with little pressure of course) - and values returned to "any results" PM10 again about 3ug - so maybe that the filters are dirty and cannot be cleaned automatically?
I have called the clean process several times before (can hear the sound) - but didn't help.

winkj, 2019-06-28:
Are you using 5V supply voltage and I2C level shifters for the ESP8266? The SPS30 is specified for 5V, whereas the ESP8266 is a 3.3V board.

michapr, 2019-06-28:
Yes, of course. And using stable 5V for SPS30.
As I said - it was working very fine since January, since March outside (before inside for testing and optimize sketch)
Now (after my "cleaning") I receive again values.

winkj, 2019-07-01:
ok, thanks for the report. I will go ahead and close this issue then, as I believe there's no issue related to the arduino-sps software

closed 2019-07-01 by winkj

michapr, 2019-07-01:
No, no issue related to software, but to device...
Is there any other place to discuss hardware issues?
Customer want to use the sensor in production - so it should be stable...
Thanks!

winkj, 2019-07-01:
There is a support team to provide help with technical questions and issues with Sensirion products. They can be reached via this form: https://www.sensirion.com/en/about-us/contact/. Alternatively, if you bought parts through a distributor, they often have support channels as well.

For any questions regarding the software, we (the Sensirion github team) are doing our best to provide timely answers, but unfortunately we're too far removed from the commercial and hardware side to go beyond that. Thanks for understanding.

Mohit2183, 2020-01-11:
we were testing your sps30 by connecting with docklight.  we have used usb to serial converter to command the sps30. pin 1 was connected to poer supply +5vdc , pin 5 was connected to 0vdc. pin4 was left open for UART, pin 2 was connected to Tx of master (PC), pin 3 was connected to Rx of master (pc). From docklight v2.3  we had set the com port , 115200, 8N1 communication settings.  we have placed command in format of Hex codes i.e. start command Tx: 7E 00 00 01 02 F9 7E , Rx REPLY: 7E 00 00 01 02 F9 7E , Tx: 7E 00 00 00 FF 7E, Rx REPLY: x: 7E 00 00 00 FF 7E , Tx: 7E 00 03 00 FC 7E ,  Rx REPLY: 7E 00 03 00 FC 7E , Tx:  7E 00 D3 00 2C 7E, Rx REPLY:  7E 00 D3 00 2C 7E .   OVER ALL FAN IS ALSO NOT ROTATING PHYSICALLY.. could you pls guide us why sps30 is not giving response to our commands.

NarinLab, 2020-03-24:
We are also getting the stable constant value after using it for 1 hour.

I'm using https://github.com/paulvha/sps30, because this one https://github.com/Sensirion/arduino-sps is not working and just keep showing probe error.

rnestler, 2020-03-25:
> We are also getting the stable constant value after using it for 1 hour.

@NarinLab Can you contact Sensirion technical support about this issue? https://www.sensirion.com/en/about-us/contact/. They have generally more knowledge about the Sensor hardware.

> I'm using https://github.com/paulvha/sps30, because this one https://github.com/Sensirion/arduino-sps is not working and just keep showing probe error.

That's unfortunate. Could you try if the version in https://github.com/Sensirion/arduino-sps/pull/16 works for you? It's hard for us to give support for a driver that isn't written by us.

rnestler, 2020-03-25:
@Mohit2183 It looks like your serial driver just echos back whatever you send. So I suspect that the sensor never actually receives any of the data you send.

Also this is the repository for the I2C driver for Arduino. If you are looking for a UART implementation take a look at https://github.com/Sensirion/embedded-uart-sps