Sensirion/arduino-avr-legacy-i2c-sps30 #15: do not compile

pat1, 2019-08-15:
avr-g++  7.4.0
avr-g++  7.3.0
avr-g++  5.4.0

avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega644p -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_PLUS -DARDUINO_ARCH_AVR -DSERIAL_TX_BUFFER_SIZE=192 -DSERIAL_RX_BUFFER_SIZE=192 -fno-strict-aliasing -std=c++11 -I/home/pat1/git/rmap/arduino/sketchbook/hardware/Microduino/avr/cores/arduino -I/home/pat1/git/rmap/arduino/sketchbook/hardware/Microduino/avr/variants/plus -I/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps /home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_common.cpp -o /tmp/arduino_build_422382/libraries/sensirion-sps/sensirion_common.cpp.o
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_common.cpp: In function 's16 sensirion_i2c_write_cmd(u8, u16)':
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_common.cpp:129:47: error: invalid conversion from 'uint16_t* {aka unsigned int*}' to 'const u16* {aka const short unsigned int*}' [-fpermissive]
     sensirion_fill_cmd_send_buf(buf, command, (uint16_t *)NULL, 0);
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_common.cpp:66:5: note:   initializing argument 3 of 'u16 sensirion_fill_cmd_send_buf(u8*, u16, const u16*, u8)'
 u16 sensirion_fill_cmd_send_buf(u8 *buf, u16 cmd, const u16 *args, u8 num_args)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_common.cpp: In function 's16 sensirion_i2c_delayed_read_cmd(u8, u16, u32, u16*, u16)':
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_common.cpp:147:43: error: invalid conversion from 'uint16_t* {aka unsigned int*}' to 'const u16* {aka const short unsigned int*}' [-fpermissive]
     sensirion_fill_cmd_send_buf(buf, cmd, (uint16_t *)NULL, 0);
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_common.cpp:66:5: note:   initializing argument 3 of 'u16 sensirion_fill_cmd_send_buf(u8*, u16, const u16*, u8)'
 u16 sensirion_fill_cmd_send_buf(u8 *buf, u16 cmd, const u16 *args, u8 num_args)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_hw_i2c_implementation.cpp: In function 's8 sensirion_i2c_write(u8, const u8*, u16)':
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_hw_i2c_implementation.cpp:68:45: error: invalid conversion from 'const u8* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
     return I2c.write(address, data[0], data + 1, count - 1);
                                        ~~~~~^~~
In file included from /home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/sensirion_hw_i2c_implementation.cpp:44:0:
/home/pat1/git/rmap/arduino/sketchbook/libraries/sensirion-sps/i2c_master_lib.h:111:13: note:   initializing argument 3 of 'uint8_t I2C::write(uint8_t, uint8_t, uint8_t*, uint8_t)'
     uint8_t write(uint8_t, uint8_t, uint8_t*, uint8_t);
             ^~~~~

fast patch:

diff -Nur sensirion-sps/sensirion_common.cpp ~/git/arduino-sps/sensirion_common.cpp 
--- sensirion-sps/sensirion_common.cpp	2019-08-15 12:22:54.301685220 +0200
+++ /home/pat1/git/arduino-sps/sensirion_common.cpp	2019-08-15 13:00:52.905324340 +0200
@@ -126,7 +126,7 @@
 s16 sensirion_i2c_write_cmd(u8 address, u16 command) {
     u8 buf[SENSIRION_COMMAND_SIZE];
 
-    sensirion_fill_cmd_send_buf(buf, command, (u16 *)NULL, 0);
+    sensirion_fill_cmd_send_buf(buf, command, (uint16_t *)NULL, 0);
     return sensirion_i2c_write(address, buf, SENSIRION_COMMAND_SIZE);
 }
 
@@ -144,7 +144,7 @@
     s16 ret;
     u8 buf[SENSIRION_COMMAND_SIZE];
 
-    sensirion_fill_cmd_send_buf(buf, cmd, (u16 *)NULL, 0);
+    sensirion_fill_cmd_send_buf(buf, cmd, (uint16_t *)NULL, 0);
     ret = sensirion_i2c_write(address, buf, SENSIRION_COMMAND_SIZE);
     if (ret != STATUS_OK)
         return ret;

diff -Nur sensirion-sps/sensirion_hw_i2c_implementation.cpp ~/git/arduino-sps/sensirion_hw_i2c_implementation.cpp 
--- sensirion-sps/sensirion_hw_i2c_implementation.cpp	2019-08-15 12:31:13.071254906 +0200
+++ /home/pat1/git/arduino-sps/sensirion_hw_i2c_implementation.cpp	2019-08-15 13:00:52.906324365 +0200
@@ -58,7 +58,7 @@
     return I2c.read(address, count, data);
 }
 
-s8 sensirion_i2c_write(u8 address, u8* data, u16 count)
+s8 sensirion_i2c_write(u8 address, const u8* data, u16 count)
 {
     // the API doesn't forsee calls without register, so we'll use the first
     // byte as "register", and pass the rest as data argument
@@ -100,7 +100,7 @@
     return 0;
 }
 
-s8 sensirion_i2c_write(u8 address, u8* data, u16 count)
+s8 sensirion_i2c_write(u8 address, const u8* data, u16 count)
 {
     Wire.beginTransmission(address);
     Wire.write(data, count);

diff -Nur sensirion-sps/sensirion_i2c.h ~/git/arduino-sps/sensirion_i2c.h 
--- sensirion-sps/sensirion_i2c.h	2019-08-15 12:32:55.158827913 +0200
+++ /home/pat1/git/arduino-sps/sensirion_i2c.h	2019-08-15 13:00:52.906324365 +0200
@@ -41,7 +41,7 @@
 
 s8 sensirion_i2c_read(u8 address, u8* data, u16 count);
 
-s8 sensirion_i2c_write(u8 address, u8* data, u16 count);
+s8 sensirion_i2c_write(u8 address, const u8* data, u16 count);
 
 void sensirion_sleep_usec(u32 useconds);

abrauchli, 2019-08-16:
Thanks for the report and the patch @pat1. As the readme states, this driver is based on https://github.com/Sensirion/embedded-sps which has been updated in the mean time. We'll address this on the fly. I'm keeping the issue open until it's resolved.

pat1, 2019-08-17:
I attach a new arduino-sps release merged and in sync with master embedded-sps
[arduino-sps.zip](https://github.com/Sensirion/arduino-sps/files/3511835/arduino-sps.zip)

One patch was needed to compile the embedded version too

pat1, 2019-08-17:
sorry the library in zip file compile but do not work :(

jucajuca, 2020-04-23:
i am experiencing the same error, is there any update for this?

abrauchli, 2020-04-23:
@jucajuca Does this version work? If not could you post your compile error?
https://github.com/Sensirion/arduino-sps/archive/update-embedded-sps-i2c.zip

pat1, 2020-04-23:
sorry I cannot help.
Now I use a patched library from https://github.com/paulvha/sps30
you can found it at:
https://github.com/r-map/rmap/tree/master/arduino/sketchbook/libraries/sps30

jucajuca, 2020-04-23:
thanks, unfortunately it did not work, see the error log below:

```

/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:61:44: error: ‘u8’ has not been declared
 int8_t sensirion_i2c_read(uint8_t address, u8* data, uint16_t count)
                                            ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp: In function ‘int8_t sensirion_i2c_read(uint8_t, int*, uint16_t)’:
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:61:68: error: conflicting declaration of C function ‘int8_t sensirion_i2c_read(uint8_t, int*, uint16_t)’
 int8_t sensirion_i2c_read(uint8_t address, u8* data, uint16_t count)
                                                                    ^
In file included from /mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:33:0:
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_i2c.h:74:8: note: previous declaration ‘int8_t sensirion_i2c_read(uint8_t, uint8_t*, uint16_t)’
 int8_t sensirion_i2c_read(uint8_t address, uint8_t *data, uint16_t count);
        ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:63:41: error: invalid conversion from ‘int*’ to ‘uint8_t {aka unsigned char}’ [-fpermissive]
     return I2c.read(address, count, data);
                                         ^
In file included from /mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:45:0:
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:114:13: note:   initializing argument 3 of ‘uint8_t I2C::read(uint8_t, uint8_t, uint8_t)’
     uint8_t read(uint8_t, uint8_t, uint8_t);
             ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp: At global scope:
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:66:51: error: ‘u8’ does not name a type
 int8_t sensirion_i2c_write(uint8_t address, const u8* data, uint16_t count)
                                                   ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp: In function ‘int8_t sensirion_i2c_write(uint8_t, const int*, uint16_t)’:
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:66:75: error: conflicting declaration of C function ‘int8_t sensirion_i2c_write(uint8_t, const int*, uint16_t)’
 int8_t sensirion_i2c_write(uint8_t address, const u8* data, uint16_t count)
                                                                           ^
In file included from /mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:33:0:
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_i2c.h:87:8: note: previous declaration ‘int8_t sensirion_i2c_write(uint8_t, const uint8_t*, uint16_t)’
 int8_t sensirion_i2c_write(uint8_t address, const uint8_t *data,
        ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:73:59: error: no matching function for call to ‘I2C::write(uint8_t&, const int&, const int*, uint16_t)’
     return I2c.write(address, data[0], data + 1, count - 1);
                                                           ^
In file included from /mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/sensirion_hw_i2c_implementation.cpp:45:0:
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:106:13: note: candidate: uint8_t I2C::write(uint8_t, uint8_t)
     uint8_t write(uint8_t, uint8_t);
             ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:106:13: note:   candidate expects 2 arguments, 4 provided
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:107:13: note: candidate: uint8_t I2C::write(int, int)
     uint8_t write(int, int); 
             ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:107:13: note:   candidate expects 2 arguments, 4 provided
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:108:13: note: candidate: uint8_t I2C::write(uint8_t, uint8_t, uint8_t)
     uint8_t write(uint8_t, uint8_t, uint8_t);
             ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:108:13: note:   candidate expects 3 arguments, 4 provided
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:109:13: note: candidate: uint8_t I2C::write(int, int, int)
     uint8_t write(int, int, int);
             ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:109:13: note:   candidate expects 3 arguments, 4 provided
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:110:13: note: candidate: uint8_t I2C::write(uint8_t, uint8_t, char*)
     uint8_t write(uint8_t, uint8_t, char*);
             ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:110:13: note:   candidate expects 3 arguments, 4 provided
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:111:13: note: candidate: uint8_t I2C::write(uint8_t, uint8_t, uint8_t*, uint8_t)
     uint8_t write(uint8_t, uint8_t, uint8_t*, uint8_t);
             ^
/mnt/FILES/CODE/Arduino/libraries/arduin_sps_i2c/i2c_master_lib.h:111:13: note:   no known conversion for argument 3 from ‘const int*’ to ‘uint8_t* {aka unsigned char*}’

```

jucajuca, 2020-04-23:
@abrauchli 

I got it working but with the code of the branch update-embedded-sps-i2c

and I had to add the following cast : /

```
int8_t sensirion_i2c_write(uint8_t address, const uint8_t* data, uint16_t count)
{
    // the API doesn't forsee calls without register, so we'll use the first
    // byte as "register", and pass the rest as data argument
    if (count == 0) {
      return 0;
    }
    uint8_t* dataPass = (uint8_t*) data;
    return I2c.write(address, data[0], dataPass + 1, count - 1);
}
```

abrauchli, 2020-04-23:
Thanks for testing and the feedback. It looks like the I2C interface is "wrong" as (IMO) it shouldn't need a non-const pointer. Casting away constness is rather dangerous but seems to be the only workaround here - also I don't expect the data to be passed any further so there shouldn't fallout from that. I'll fix it in the pull-request.

abrauchli, 2020-04-23:
@jucajuca https://github.com/Sensirion/arduino-sps/archive/update-embedded-sps-i2c.zip could you try again and see if it also fixes the other warnings?

jucajuca, 2020-04-23:
yup thanks, it is working on arduino duemilanove

abrauchli, 2020-04-24:
@jucajuca thanks again for testing, I'm closing this as the change was merged to master. Feel free to open a new issue if something doesn't work.

closed 2020-04-24 by abrauchli