Sensirion/arduino-avr-legacy-i2c-sps30 #18: Are I2C level shifters really needed for 3.3V devices?

Tojaj, 2020-01-12:
Hi, the doc states:

> When using a 3.3V based Arduino, make sure to use the appropriate voltage regulators and level shifters for I2C!

I wonder if the need for I2C level shifters is real for SPS30?

* If a 3,3V device is acting like an I2C master, then the I2C bus should be also driven on 3,3V.
* SPS30 data sheet [1], in "Hardware Interface Specifications" chapter, states that I2C SDA/SCL are "TTL 5V and LVTTL 3.3V compatible" = SPS30 is fine with 3,3V on I2C.

That said, in case of SPS30 sensor and 3,3V devices, one only need to make sure to provide an appropriate voltage 5V (±10%) to the sensor.

Btw. Some NodeMCU ESP8266 devices provides 5V directly from powering USB on its VU/VIN pin. For such NodeMCU devices, it should be possible to use the SPS30 sensor directly without any level shifter (for I2C) or step-up converter (for sensor power supply).

[1] https://cdn.sparkfun.com/assets/2/d/2/a/6/Sensirion_SPS30_Particulate_Matter_Sensor_v0.9_D1__1_.pdf

abrauchli, 2020-01-14:
Hi @Tojaj, while it's not really a driver question, I do think it may be of relevance to others. Let me see what I can find out..

abrauchli, 2020-01-14:
Hi @Tojaj 

So you're right that you don't need the level shifters. Just make sure that the 5V supply can provide 80mA peak (at startup).
Both informations should become part of the next datasheet update but I can't give you an ETA.
The most recent datasheet is available here: https://www.sensirion.com/file/datasheet_sps30

Cheers

closed 2020-01-14 by abrauchli

winkj, 2020-01-14:
I prepared an update for the README too, this should be clarified soon. Thanks @Tojaj for the report

Tojaj, 2020-01-14:
Thank you both! @abrauchli @winkj

SebastianKotstein, 2022-07-14:
How does the wiring in case of a 3.3V I2C look like? Is it correct that the SPS30 must be powered with 5V via VDD (Pin 1), but SCL and SDA must be pulled up to 3.3V (rather than VDD / 5V as illustrated in [Figure 8 in the datasheet](https://cdn.sparkfun.com/assets/2/d/2/a/6/Sensirion_SPS30_Particulate_Matter_Sensor_v0.9_D1__1_.pdf))?

pklosko, 2022-07-17:
Hello,
5V (VDD) is necessary for internal laser detector.
In https://github.com/pklosko/AirQmonitor I have SCL&SDA pullups to 5V - because of 5V MCU GPIO levels