Sensirion/arduino-avr-legacy-i2c-sps30 #20: SPS30 unexpected readings

dominik-cnx, 2020-07-28:
I'm running

https://github.com/Sensirion/arduino-sps/blob/master/examples/sps30/sps30.ino (unchanged)

on a ESP32 and have the SPS30 connected through I2C. This are the results I am getting

SPS sensor probing successful
measurements started
PM  1.0: 7608.96
PM  2.5: 8046.18
PM  4.0: 8046.19
PM 10.0: 8046.18
NC  0.5: 53145.59
NC  1.0: 60563.71
NC  2.5: 60725.93
NC  4.0: 60746.30
NC 10.0: 60756.48
Typical partical size: 0.28

PM  1.0: 6177.64
PM  2.5: 6532.62
PM  4.0: 6532.63
PM 10.0: 6532.62
NC  0.5: 43148.42
NC  1.0: 49171.13
NC  2.5: 49302.83
NC  4.0: 49319.37
NC 10.0: 49327.64
Typical partical size: 0.28

I have a couple of PM monitors right next to the SPS30 and they give me a PM2.5 value very close to 0 (some actually 0).

So we have 

PM  2.5: 8046.18 measured by the SPS30 

vs 

0 ug measured by my reference devices.

I tested two devices and the result is the same.

winkj, 2020-07-28:
Just to make sure, are you running the SPS30 at 5V?

dominik-cnx, 2020-07-28:
Thx, no 3.3V. Had a look here

https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9.6_Particulate_Matter/Datasheets/Sensirion_PM_Sensors_SPS30_Datasheet.pdf

Looks like it wants 5V. Will feed it 5V and report back.

winkj, 2020-07-28:
I've ran it on 3.3v at some points and got values which where about 1000x higher than at 5V, so there's a chance this will get you in the right order of magnitude

dominik-cnx, 2020-07-28:
Looks fine with 5V, many thanks!

PM  1.0: 0.69
PM  2.5: 0.75
PM  4.0: 0.78
PM 10.0: 0.78
NC  0.5: 4.66
NC  1.0: 5.42
NC  2.5: 5.47
NC  4.0: 5.48
NC 10.0: 5.48
Typical partical size: 0.64

closed 2020-07-28 by dominik-cnx