Sensirion/arduino-avr-legacy-i2c-sps30 #23: Feature request: Allow using secondary Wire buses

matthijskooijman, 2020-10-31:
On some boards, such as STM32 boards (and probably SAMD boards too), there are multiple I²C controllers available, typically named `Wire`, `Wire1`, etc.

Currently, this library always uses `Wire` hardcoded, but it would be convenient if a secondary bus could also be used.

There is already a `sensirion_i2c_select_bus(uint8_t idx)` function declared, but it is never defined anywhere. Also, since I'm not sure whether there is a standard way to detect how much Wire objects are defined, this API might be hard to implement on Arduino. Therefore, I'd suggest implementing a selection function that actually passes the `Wire`, `Wire1`, etc objects as a reference. The type of these seems to be `TwoWire` on most cores, but just in case cores use other names, I'd suggest using `decltype(Wire)` instead.

So, that would be something like this (untested, should probably be split between .h and .cpp files as well, but just to get the general idea):

```
using SensirionWireType = decltype(Wire);

SensirionWireType *SensirionWire = &Wire;

sensirion_i2c_select_bus(SensirionWireType& bus)
{
  SensirionWire = &bus;
}
```

And then replace all other `Wire.` with `SensirionWire->` too.

This allows a sketch to do e.g.:

```
sensirion_i2c_select_bus(Wire1);
sensirion_i2c_init();
```

It can even switch between multiple buses at runtime, as long as it does call `sensiron_i2c_init()` once for each bus beforehand.

winkj, 2020-10-31:
Sounds good; this is partly because all the interfaces in sensirion_hw_i2c_implementation.cpp use generic - i.e. no platform specific - APIs which is why we had the sensirion_i2c_select_bus() call in there with an index; changing the interface to an Arduino SDK one would break compilation on other platforms.

So this will require a little bit more thought on our end, or potentially just forgoing compatibility with embedded-common in the interest of better suppport of Arduino platform needs

matthijskooijman, 2020-10-31:
>  changing the interface to an Arduino SDK one would break compilation on other platforms.

Yeah, so maybe *adding* a function (i.e. `sensirion_i2c_select_wire(...)` or so) could work? And then just ignoring the existing `sensirion_i2c_select_bus()`?

Even though the existing index-based API isn't necessarily bad, I'm afraid it will be hard to implement fully portably Arduino, since there is no generic index -> Wire object API, or a standardized and documented way to figure out how many wire interfaces there are.

Looking more closely, it does seem that there is a `WIRE_INTERFACES_COUNT` macro that is at least used by the Arduino `sam`, Arduino `samd`, Adafruit `samd`, `stm32l0` and `stm32l4` cores, so that could be a reasonably standard way to do this maybe? It seems that Teensy implements `WIRE_IMPLEMENT_WIREX` macros instead. So an alternative could be maybe:

```
int16_t sensirion_i2c_select_bus(uint8_t bus_idx) {
  switch (bus_idx) {
    case 0:
      SensirionWire = &Wire;
      return 0;
#if defined WIRE_IMPLEMENT_WIRE1 || WIRE_INTERFACES_COUNT > 1
    case 1:
      SensirionWire = &Wire1;
      return 0;
#endif
#if defined WIRE_IMPLEMENT_WIRE2 || WIRE_INTERFACES_COUNT > 2
    case 1:
      SensirionWire = &Wire2;
      return 0;
#endif
#if defined WIRE_IMPLEMENT_WIRE3 || WIRE_INTERFACES_COUNT > 3
    case 1:
      SensirionWire = &Wire3;
      return 0;
#endif
    default:
      return STATUS_FAIL;
  }
}
```

(as an alternative implementation for `sensirion_i2c_select_bus`, the other changes suggested above are still needed)

abrauchli, 2020-11-01:
The generic usage we thought of is quite similar but shifts more logic in to the i2c read and write functions based on the current context. A quick recap: The HAL implementation is aware of all three implementations: `sensirion_i2c_read`, `sensirion_i2c_write` and `sensirion_i2c_select_bus` - the latter being the trigger to switching the context (since we don't pass context objects, those have to be maintained separate, e.g. as globals.) Thus, we modify the behavior of `sensirion_i2c_read` depending on the context, which is triggered by the index. E.g. the read function would then look like this `switch (index) { case 1: Wire1.read(); // ...`

matthijskooijman, 2020-11-01:
Ah, that would probably be even better, since then the code will even work when `Wire` and `Wire1` have different types somehow.

winkj, 2023-09-20:
The cleanest way forward would be to port this to https://github.com/Sensirion/arduino-core/ since that allows selecting Wire objects already.

In the interest of time, maybe we could just modify `sensiron_i2c_init()` to take a reference to an initialized Wire object... @psachs, thoughts?

matthijskooijman, 2024-08-12:
Any further thoughts on this issue? Would be good if we could switch back to an unmodified arduino-sps library for our project.

psachs, 2024-10-28:
If it helps you in any way. We already have a newer driver for SCD30 that allows you to provide the Wire and I2C address on initialization. The driver is available on Arduino library and PlatformIO.

https://github.com/Sensirion/arduino-i2c-scd30

psachs, 2025-02-18:
@matthijskooijman yes sorry for my wrong suggestion. This driver is for a different sensor and I somehow managed to reference the wrong driver.

Since we are not planning to release a driver for sps30 with the new interface soon, I think the easiest solution is what @winkj suggested to allow passing of an initialized Wire object .

matthijskooijman, 2025-02-18:
Thanks for the confirmation. I've prepared a PR that implements this suggestion, see #54. Would be nice if that could be merged and released, then I can revert back to the default version of this library (and not have our users run into trouble if they accidentally reverted to the default version using the Arduino library manager).

closed 2025-02-19 by psachs

g6094199, 2025-03-17:
despite the assumption from @matthijskooijman that the [upgrade](https://github.com/Sensirion/arduino-sps/pull/54) from 1.1.2 to 1.2 

> This change should be fully backward compatible

it isnt. at least on SAMD (Arduino MKRWIFI) i get the following error:

`ino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8054 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR WiFi 1010\"" -DUSE_BQ24195L_PMIC -I/home/swirthmann/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/home/swirthmann/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/cores/arduino/api/deprecated -I/home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/cores/arduino/api/deprecated-avr-comp -I/home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/cores/arduino -I/home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/variants/mkrwifi1010 -I/home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/libraries/SPI -I/home/swirthmann/Arduino/libraries/WiFiNINA/src -I/home/swirthmann/Arduino/libraries/Arduino_MKRENV/src -I/home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/libraries/Wire -I/home/swirthmann/Arduino/libraries/NTP -I/home/swirthmann/Arduino/libraries/SD/src -I/home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/libraries/SDU/src -I/home/swirthmann/Arduino/libraries/ArduinoOTA/src -I/home/swirthmann/Arduino/libraries/Uptime/src -I/home/swirthmann/Arduino/libraries/RTClib/src -I/home/swirthmann/Arduino/libraries/Adafruit_BusIO -I/home/swirthmann/Arduino/libraries/RTCZero/src -I/home/swirthmann/Arduino/libraries/SparkFun_SCD30_Arduino_Library/src -I/home/swirthmann/Arduino/libraries/Adafruit_SGP40_Sensor/src -I/home/swirthmann/Arduino/libraries/Sensirion_Core/src -I/home/swirthmann/Arduino/libraries/Sensirion_I2C_SFA3x/src -I/home/swirthmann/Arduino/libraries/arduino-sht -I/home/swirthmann/Arduino/libraries/sensirion-sps -I/home/swirthmann/Arduino/libraries/ArduinoJson/src -I/home/swirthmann/Arduino/libraries/ArduinoMqttClient/src -I/home/swirthmann/Arduino/libraries/StreamUtils/src /home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_common.cpp -o /home/swirthmann/.var/app/cc.arduino.IDE2/cache/arduino/sketches/9494CF4DBCA828AA997FFCC531472133/libraries/sensirion-sps/sensirion_common.cpp.o
In file included from /home/swirthmann/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21.h:69:0,
                 from /home/swirthmann/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd.h:105,
                 from /home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/cores/arduino/WVariant.h:22,
                 from /home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/variants/mkrwifi1010/variant.h:24,
                 from /home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/libraries/Wire/Wire.h:24,
                 from /home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_i2c.h:37,
                 from /home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_hw_i2c_implementation.cpp:33:
/home/swirthmann/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21g18a.h:226:0: warning: "LITTLE_ENDIAN" redefined
 #define LITTLE_ENDIAN          1
 
In file included from /home/swirthmann/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/sys/types.h:67:0,
                 from /home/swirthmann/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/stdio.h:61,
                 from /home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/cores/arduino/api/Print.h:22,
                 from /home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/cores/arduino/api/Stream.h:25,
                 from /home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/cores/arduino/api/HardwareI2C.h:22,
                 from /home/swirthmann/.arduino15/packages/arduino/hardware/samd/1.8.14/libraries/Wire/Wire.h:23,
                 from /home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_i2c.h:37,
                 from /home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_hw_i2c_implementation.cpp:33:
/home/swirthmann/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/machine/endian.h:17:0: note: this is the location of the previous definition
 #define LITTLE_ENDIAN _LITTLE_ENDIAN
 
In file included from /home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_hw_i2c_implementation.cpp:33:0:
/home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_i2c.h:46:25: error: variable or field 'sensirion_i2c_init' declared void
 void sensirion_i2c_init(TwoWire& wire);
                         ^~~~~~~
/home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_i2c.h:46:25: error: 'TwoWire' was not declared in this scope
/home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_i2c.h:46:25: note: suggested alternative:
[... cut ...]
                 from /home/swirthmann/Arduino/libraries/sensirion-sps/sps30.cpp:32:
/home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_i2c.h:46:34: error: 'wire' was not declared in this scope
 void sensirion_i2c_init(TwoWire& wire);
                                  ^~~~
/home/swirthmann/Arduino/libraries/sensirion-sps/sensirion_i2c.h:46:34: note: suggested alternative: 'Wire'
 void sensirion_i2c_init(TwoWire& wire);
                                  ^~~~
                                  Wire`

matthijskooijman, 2025-03-17:
Ah, good point. I checked for the class name `TwoWire`, which is indeed defined, but the SAMD core puts it into an `arduino` namespace: https://github.com/arduino/ArduinoCore-samd/blob/993398cb7a23a4e0f821a73501ae98053773165b/libraries/Wire/Wire.h#L30-L32

This would be easy to fix om SAMD with `using arduino::TwoWire;`, but that breaks on platforms that do not have this namespace, so that is not a solution.

Before my original PR, I did already test a solution where the TwoWire type is autodetected using `decltype`. I've just added this and tested on SAMD, AVR, ESP8266, ESP32 and STM32L0, which all seems to work (compile) with this PR: https://github.com/Sensirion/arduino-sps/pull/56

commit 835a4b58 (2025-02-19): Allow passing secondary wire buses to sensirion_i2c_init()

This allows using other I2C/Wire buses than the first one, e.g.:

    sensirion_i2c_init(Wire1);

The existing function (without arguments) is still availble, so existing
code will just default to Wire just as before.

This adds a second function instead of using a default argument value,
so the original function (with C-linkage) can remain, allowing this
library to also still be usable from C-code (but specifying a secondary
Wire bus is only supported from C++ for obvious reasons).

This fixes #23.