Sensirion/arduino-avr-legacy-i2c-sps30 #25: Suggestion: Autogenerate arduino library

matthijskooijman, 2020-11-01:
This library contains code from the https://github.com/Sensirion/embedded-sps repository, with some additions to turn it into an Arduino library. I suspect this is done by manually copying the code from that rep into this one manually every now and then, which means the code is being tracked in multiple places and development is not particularly easy.

For another project, I had the same challenge (making an upstream, non-Arduino specific library work on Arduino) and I chose a different approach: Include the Arduino-specific files in a subdirectory of the generic repository and include a small script that automatically copies and reorders the needed files into an Arduino library directory. This allows all development to happen in one repository, and just occasionally running the script (manually, or maybe automatically using a github action) to publish the generated arduino library in a separate repository. By using symlinks rather than copies during development, making changes is even easier (no need to rerun the script after every change).

Maybe this approach would also be useful for this library?

The script I'm talking about is here: https://github.com/LacunaSpace/basicmac/blob/master/target/arduino/export.sh (there's also a Windows version being worked on in the PRs).

rnestler, 2020-11-02:
Hi @matthijskooijman 
Thanks for the feedback. We are actually working on generating Arduino drivers automatically instead of maintaining them manually. I expect first results by the end of this year and hopefully we can automatically generate Arduino drivers for all our sensor by Q2 of 2021.

rnestler, 2021-01-29:
We started automatically generating Arduino drivers: https://github.com/sensirion/arduino-i2c-scd4x. Note that the actual code is generated as well.

MBjoern, 2021-04-06:
As there was no activity on this issue, i'll close it. Feel free to reopen or open another issue, if you need any further assistance.

Keep prototyping, We'll love to see your projects in the field.

closed 2021-04-06 by MBjoern