Sensirion/arduino-avr-legacy-i2c-sps30 #28: Read sps30 in esp32 with other wires

mourazo, 2021-01-15:
I can't read with ESP32 with other GPIO's, I try to begin wire with Wire.begin(25,26) but don't pass the probe sps30_probe(). I can't find a method in the library for assign other GPIO's to scl and sda

psachs, 2021-01-15:
Hi

On ESP32 the default hardware I2C pins are used which are 21 (SDA) and 22 (SCL). 

Did you change the Wire.begin() in the following file? That is where the Wire is initialized for this library
https://github.com/Sensirion/arduino-sps/blob/master/sensirion_hw_i2c_implementation.cpp#L86

Best regards,
Pascal

mourazo, 2021-01-15:
First of all thanks for your quick answer, I updated what you say but I recive the nexts errors in the middle of measurments.

```
18:06:15.361 -> error reading measurement
18:06:16.346 -> error reading measurement
18:06:17.348 -> error reading measurement
18:06:18.348 -> error reading measurement
18:06:19.350 -> PM  1.0: 5.90
18:06:19.350 -> PM  2.5: 6.24
18:06:19.392 -> PM  4.0: 6.24
18:06:19.392 -> PM 10.0: 6.24
18:06:19.392 -> NC  0.5: 41.02
18:06:19.392 -> NC  1.0: 46.95
18:06:19.392 -> NC  2.5: 47.08
18:06:19.392 -> NC  4.0: 47.10
18:06:19.392 -> NC 10.0: 47.11
18:06:19.392 -> Typical partical size: 0.55
18:06:19.392 -> 
18:06:20.390 -> error reading measurement
18:06:21.382 -> error reading measurement
18:06:22.490 -> error reading data-ready flag: -1
18:06:22.592 -> error reading measurement
18:06:23.604 -> error reading measurement
18:06:24.607 -> error reading measurement
18:06:25.611 -> error reading measurement
18:06:26.617 -> error reading measurement
18:06:27.605 -> error reading data-ready flag: -1
18:06:27.732 -> error reading measurement
18:06:28.739 -> PM  1.0: 5.74
18:06:28.739 -> PM  2.5: 6.07
18:06:28.739 -> PM  4.0: 6.07
18:06:28.739 -> PM 10.0: 6.07
18:06:28.739 -> NC  0.5: 39.89
18:06:28.739 -> NC  1.0: 45.67
18:06:28.739 -> NC  2.5: 45.79
18:06:28.739 -> NC  4.0: 45.81
18:06:28.739 -> NC 10.0: 45.82
18:06:28.739 -> Typical partical size: 0.54
18:06:28.739 -> 
18:06:29.823 -> error reading measurement
18:06:30.796 -> error reading measurement

```

winkj, 2021-01-15:
Weird that it seems to work just fine occasionally... do you have any other I2C devices on the same bus? And do you know what speed the bus is running at?

mourazo, 2021-01-15:
Yes, I have a sensirion scd30 on the same bus, but it works correctly with the two connected devices, I only have an error with the sps30, if I read in the default GPIO scl and sda of the esp32 I don't have the error that I described, and I can read the two devices correctly at the same time. I only have this error in GPIO's 25 and 26. How can I know the speed of the bus?  Thanks.

winkj, 2021-01-15:
Interesting that it's working on the default pins; have you tried other pins that 25 and 26?

Regarding the bus speed: The SPS30 can do 100kHz bus speed at most, so if any of the other libraries changes the speed to 400kHz that could cause issues. Since it's working on the default pins, it's less likely that it's a bus speed issue, though.

Does this happen with our unmodified SPS30 samples, with only the SPS30 attached?

mourazo, 2021-01-16:
I checked my cable connections and now it works perfectly, the error was a cable that wasn't well in contact. Thanks.

closed 2021-01-16 by mourazo