Sensirion/arduino-avr-legacy-i2c-sps30 #29: Read issue with esp32

mourazo, 2021-01-19:
After close the previous issue https://github.com/Sensirion/arduino-sps/issues/28
And start reading with the two devices (scd30 and sps30) conected in the same bus without error measurments, I could observe that the measurments of pm2.5, pm4 and pm10 it's the same forever, that this is imposible and not the same if I measure with Sensiron ControCenter, for other hand, I get the next warning when I compile the code for esp32 with Arduino IDE:

> In file included from C:\Users\alvar\Documents\Arduino\libraries\sensirion-sps\sensirion_hw_i2c_implementation.cpp:78:0:
> C:\Users\alvar\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src/Wire.h: In function 'int8_t sensirion_i2c_read(uint8_t, uint8_t*, uint16_t)':
> C:\Users\alvar\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src/Wire.h:103:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)
>      uint8_t requestFrom(int address, int size);
>              ^
> C:\Users\alvar\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src/Wire.h:101:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)
>      uint8_t requestFrom(uint8_t address, uint8_t size);

I think the PM measurements are mixed up and I can only see one of the 3 PM measurements, I get the error with all the sketches I test, like this:

```
#include <sps30.h>
#include <Wire.h>
// Example arduino sketch, based on 
// https://github.com/Sensirion/embedded-sps/blob/master/sps30-i2c/sps30_example_usage.c

// uncomment the next line to use the serial plotter
// #define PLOTTER_FORMAT
#include "SparkFun_SCD30_Arduino_Library.h" //Click here to get the library: http://librarymanager/All#SparkFun_SCD30

SCD30 airSensor;
void setup() {
  int16_t ret;
  uint8_t auto_clean_days = 4;
  uint32_t auto_clean;

  
  Serial.begin(115200);

  delay(2000);
  Wire.begin(25,26);
   while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB
  }
  sensirion_i2c_init();

  while (sps30_probe() != 0) {
    Serial.print("SPS sensor probing failed\n");
    delay(500);
  }

#ifndef PLOTTER_FORMAT
  Serial.print("SPS sensor probing successful\n");
#endif /* PLOTTER_FORMAT */

  ret = sps30_set_fan_auto_cleaning_interval_days(auto_clean_days);
  if (ret) {
    Serial.print("error setting the auto-clean interval: ");
    Serial.println(ret);
  }

  ret = sps30_start_measurement();
  if (ret < 0) {
    Serial.print("error starting measurement\n");
  }

#ifndef PLOTTER_FORMAT
  Serial.print("measurements started\n");
#endif /* PLOTTER_FORMAT */

#ifdef SPS30_LIMITED_I2C_BUFFER_SIZE
  Serial.print("Your Arduino hardware has a limitation that only\n");
  Serial.print("  allows reading the mass concentrations. For more\n");
  Serial.print("  information, please check\n");
  Serial.print("  https://github.com/Sensirion/arduino-sps#esp8266-partial-legacy-support\n");
  Serial.print("\n");
  delay(2000);
#endif

 if (airSensor.begin() == false)
  {
    Serial.println("Air sensor not detected. Please check wiring. Freezing...");
    while (1)
      ;
  }

  delay(1000);
}

void loop() {
  struct sps30_measurement m;
  char serial[SPS30_MAX_SERIAL_LEN];
  uint16_t data_ready;
  int16_t ret;

  do {
    ret = sps30_read_data_ready(&data_ready);
    if (ret < 0) {
      Serial.print("error reading data-ready flag: ");
      Serial.println(ret);
    } else if (!data_ready)
      Serial.print("data not ready, no new measurement available\n");
    else
      break;
    delay(100); /* retry in 100ms */
  } while (1);

  ret = sps30_read_measurement(&m);
  if (ret < 0) {
    Serial.print("error reading measurement\n");
  } else {

#ifndef PLOTTER_FORMAT
    Serial.print("PM  1.0: ");
    Serial.println(m.mc_1p0);
    Serial.print("PM  2.5: ");
    Serial.println(m.mc_2p5);
    Serial.print("PM  4.0: ");
    Serial.println(m.mc_4p0);
    Serial.print("PM 10.0: ");
    Serial.println(m.mc_10p0);

#ifndef SPS30_LIMITED_I2C_BUFFER_SIZE
    Serial.print("NC  0.5: ");
    Serial.println(m.nc_0p5);
    Serial.print("NC  1.0: ");
    Serial.println(m.nc_1p0);
    Serial.print("NC  2.5: ");
    Serial.println(m.nc_2p5);
    Serial.print("NC  4.0: ");
    Serial.println(m.nc_4p0);
    Serial.print("NC 10.0: ");
    Serial.println(m.nc_10p0);

    Serial.print("Typical partical size: ");
    Serial.println(m.typical_particle_size);
#endif

    Serial.println();

#else
    // since all values include particles smaller than X, if we want to create buckets we 
    // need to subtract the smaller particle count. 
    // This will create buckets (all values in micro meters):
    // - particles        <= 0,5
    // - particles > 0.5, <= 1
    // - particles > 1,   <= 2.5
    // - particles > 2.5, <= 4
    // - particles > 4,   <= 10

    Serial.print(m.nc_0p5);
    Serial.print(" ");
    Serial.print(m.nc_1p0  - m.nc_0p5);
    Serial.print(" ");
    Serial.print(m.nc_2p5  - m.nc_1p0);
    Serial.print(" ");
    Serial.print(m.nc_4p0  - m.nc_2p5);
    Serial.print(" ");
    Serial.print(m.nc_10p0 - m.nc_4p0);
    Serial.println();

#endif /* PLOTTER_FORMAT */

  }

delay(1000);

if (airSensor.dataAvailable())
  {
    Serial.print("co2(ppm):");
    Serial.print(airSensor.getCO2());

    Serial.print(" temp(C):");
    Serial.print(airSensor.getTemperature(), 1);

    Serial.print(" humidity(%):");
    Serial.print(airSensor.getHumidity(), 1);

    Serial.println();
  }
  else
    Serial.println("Waiting for new data");
  delay(2000);
}
```

Example of output: 

> 13:38:14.623 -> co2(ppm):914 temp(C):25.2 humidity(%):24.7
> 13:38:16.632 -> PM  1.0: 3.21
> 13:38:16.632 -> PM  2.5: 3.39
> 13:38:16.632 -> PM  4.0: 3.39
> 13:38:16.632 -> PM 10.0: 3.39
> 13:38:16.632 -> NC  0.5: 22.31
> 13:38:16.632 -> NC  1.0: 25.53
> 13:38:16.667 -> NC  2.5: 25.60
> 13:38:16.667 -> NC  4.0: 25.61
> 13:38:16.667 -> NC 10.0: 25.62
> 13:38:16.667 -> Typical partical size: 0.52
> 13:38:16.667 -> 
> 13:38:17.634 -> co2(ppm):914 temp(C):25.2 humidity(%):24.7
> 13:38:19.671 -> PM  1.0: 3.29
> 13:38:19.671 -> PM  2.5: 3.48
> 13:38:19.671 -> PM  4.0: 3.48
> 13:38:19.671 -> PM 10.0: 3.48
> 13:38:19.671 -> NC  0.5: 22.87
> 13:38:19.671 -> NC  1.0: 26.18
> 13:38:19.671 -> NC  2.5: 26.25
> 13:38:19.671 -> NC  4.0: 26.26
> 13:38:19.671 -> NC 10.0: 26.27
> 13:38:19.671 -> Typical partical size: 0.52
> 13:38:19.671 -> 
> 13:38:20.668 -> co2(ppm):913 temp(C):25.2 humidity(%):24.7
> 13:38:22.686 -> PM  1.0: 3.33
> 13:38:22.686 -> PM  2.5: 3.52
> 13:38:22.686 -> PM  4.0: 3.52
> 13:38:22.686 -> PM 10.0: 3.52
> 13:38:22.686 -> NC  0.5: 23.15
> 13:38:22.686 -> NC  1.0: 26.49
> 13:38:22.686 -> NC  2.5: 26.57
> 13:38:22.686 -> NC  4.0: 26.58
> 13:38:22.686 -> NC 10.0: 26.58
> 13:38:22.686 -> Typical partical size: 0.52
> 13:38:22.686 -> 
>

winkj, 2021-01-19:
The output you're showing looks fine; please remember that PMx means "all particles of 4um and smaller", so PM2.5 is a subset of PM4, and PM2.5 and PM4 are a subset of PM10. If you have mainly PM2.5 particles, the three values will therefore show the same (given the limited resolution of the output). You can see in the number concentration that there are slight differences in between PM2.5 and PM4. See issues #27 for a more detailed explanation

Could you share a screenshot of ControlCenter?

mourazo, 2021-01-23:
Sorry for my late answer, after a few days of dataloggin I can see that it is not the same data, only when there are very few particles, pm 2.5, pm 4 and pm10 they match. Sorry for the issue.

mourazo, 2021-06-07:
After testing the device for a few days I have the same problem, the values pm2.5, pm4 and pm10 always seem the same, as in my previous issue, which is impossible, can it be a problem with the code or is it possible that the problem with the firmware or laser? I am testing with three different sensors and I have the same results. Find attached the log of a few days 

[particulasTest.txt](https://github.com/Sensirion/arduino-sps/files/6608194/particulasTest.txt)

michapr, 2021-06-07:
Have you ever tried to move a match near the sensor and then blow it out?
Then the values would have to change very strongly - and different values should then also occur for PM2.5 - PM10.

mourazo, 2021-06-07:
Yes I tried and the values change very differently in the different ranges, but I am testing the SPS30 with three beta gauge reference analyzers and the values for this are greater than 10 ug/m3 while the SPS30 does not exceed 3 ug/m3. On the other hand, the values of the beta gauge reference analyzers change as soon as you enter the room, or you shake off your shirt but the values of SPS30 take his time to up and have the same in any ranges. It's so weird, something is wrong

winkj, 2021-06-07:
> After testing the device for a few days I have the same problem, the values pm2.5, pm4 and pm10 always seem the same, as in my previous issue, which is impossible [...]

As mentioned previously, this is not impossible, since PM4 includes all particles from PM2.5, and PM10 includes all particles from PM4 (and thus PM2.5). So if you primarily have small particles in your test sample, those values will be similar, or the same if you only print limited.

Here's an example for the first line of your log file:
[image]
PM2.5 would be 3.1
PM4 would be 3.2 (3.1 + 0.1)
PM10 would be 3.2

Let me know if you have any questions on this

winkj, 2021-06-07:
> Yes I tried and the values change very differently in the different ranges, but I am testing the SPS30 with three beta gauge reference analyzers [...]

If you believe that the SPS30 hardware is not working as described in the datasheet, please reach out to Sensirion's support team. Unfortunately, we can only support software issues on github, and there's nothing in your data that would suggest a driver malfunction.

Sensirion support can be reached via https://www.sensirion.com/en/about-us/contact/

closed 2021-06-07 by winkj