Sensirion/arduino-avr-legacy-i2c-sps30 #30: SPS30 is always returning 0.00 and data ready flag is not setting

Chandu-52, 2021-01-23:
I starting using SPS30 to measure PM values. But Im not exactly understanding why it is always returning 0.00ug/m3 always.
My steps were
1. Start measurement with 0x05 output format
2. Set pointer for data-ready flag
3.  In while loop - "read values using 0x3000 Address pointer" and printing it

This is how it is 
[image]

rnestler, 2021-01-26:
Hi @Chandu-52 
Are you using this driver with an Arduino or custom code? I'm not sure I follow your description on what you do are doing. You'll need to repeatably write the read measurement command (0x0300) and then read the values.

Do you get any NACKs on the bus? Are the CRC checksums correct which you receive?

winkj, 2021-01-26:
To add to @rnestler's request: can you try to reproduce the problem with our Arduino examples? That will help us narrow down the cause

Chandu-52, 2021-01-27:
> Hi @Chandu-52
> Are you using this driver with an Arduino or custom code? I'm not sure I follow your description on what you do are doing. You'll need to repeatably write the read measurement command [...]

I interfaced sensor with STM32 discovery board.
though I repeatedly read measurement commands it is responding with 0

Thanks and regards

winkj, 2021-01-27:
@Chandu-52 just to clarify: are you using the arduino-sps library in your project? If yes, do our samples work?

winkj, 2021-02-08:
Please reopen if the problem exists in our samples

closed 2021-02-08 by winkj