Sensirion/arduino-avr-legacy-i2c-sps30 #31: SPS30 always returning equal PM2.5 and PM10 concentrations

ishan2109, 2021-02-08:
The PM2.5 and PM10 mass concentration values returned by SPS30 are always in a very close range(difference <=1 ug/m3, screenshot attached below).I have used the code directly.

I am using NodeMCUv1.0 and an external power supply 5V 2A just to make sure there is no issue with power.
I also have  SDS011 sensor by Nova Fitness and Aeroqual. Both of them have are returning PM2.5 and PM10 values with a significant difference (at least 5-10 ug/m3). 

I am not sure if there is a problem with sensor or I need to do some modifications in the code itself?

[image]

rnestler, 2021-02-08:
Hi @ishan2109
I don't see anything wrong that would point in the direction of any issue with the code itself.
Since we can't really give support here regarding the sensors performance, I recommend you to contact support via https://www.sensirion.com/en/about-us/contact/

winkj, 2021-02-08:
Duplicate of #27

closed 2021-02-08 by winkj