Sensirion/arduino-avr-legacy-i2c-sps30 #34: Wrong values

mourazo, 2021-07-28:
I'm trying to read one sps30 with scd30 at the same time with an ESP32, previusly I have been able without problems, but now my sps30 returs values to big, how is it possible? Could there be problems with my wire? Thanks

`void setup()
{

    pinMode(15, OUTPUT);
    digitalWrite(15, LOW);
    delay(1);
    Serial.begin(115200); /* prepare for possible serial debug */

    ledcSetup(channel, freq, resolution);
    ledcAttachPin(21, channel);
  
    delay(100);
    Wire.begin(25,26);

        if (airSensor.begin(Wire, false) == false)
    {
        Serial.println("Air sensor not detected. Please check wiring. Freezing...");
        while (1)
            ;
    }
    
    airSensor.setMeasurementInterval(2);

  particles();

   if (!bmp.begin()) {
          Serial.println(F(" Hay algún problema con el sensor de presión!"));
          while (1) delay(10);
         }
 
   if(!SPIFFS.begin()){ 
    SPIFFS.format();
    Serial.println(" No se ha podido acceder a la memoria");  
    alarma1 = "800";
    alarma2 = "1000";
    Serial.println(" Offset establecido en 8.8");
    airSensor.setTemperatureOffset(8.8);
     airSensor.setAltitudeCompensation(690); 
    String pp = p + "123"+ p2 +"%";
    const char* ppp = pp.c_str();
    WiFi.begin("taller_seq",ppp);
    }

    if(SPIFFS.exists("/alarmas.txt")){
      Serial.println(" Alarmas cargadas");
    }
    else if(!SPIFFS.exists("/alarmas.txt")){
      fileAlarms = SPIFFS.open("/alarmas.txt", "a+");
        fileAlarms.println("800");
        fileAlarms.println("1000");
        fileAlarms.close();
    }
    
      int line=0;
      int index=0;
      fileAlarms = SPIFFS.open("/alarmas.txt", FILE_READ);
      
    while (fileAlarms.available()) {
        if(line==0){
        alarma1=fileAlarms.readStringUntil('\n');
        alarma1=alarma1.c_str();
        Serial.print(" Alarma 1: ");
        Serial.println(alarma1);
        }
        if(line==1){
        alarma2=fileAlarms.readStringUntil('\n');
        alarma2=alarma2.c_str();
        Serial.print(" Alarma 2: ");
        Serial.println(alarma2);
        }
        
        line++;
        }

        
     fileAlarms.close();

     if(SPIFFS.exists("/reconexion.txt")){
      Serial.println(" Registro de reconexion encontrado");
    }
    else if(!SPIFFS.exists("/reconexion.txt")){
      fileRec = SPIFFS.open("/reconexion.txt", "a+");
        fileRec.println("0");
        fileRec.close();
      Serial.println(" Reconexion 0");
    }
    
     mac = WiFi.macAddress();
     
     
     lv_init();

    
#if USE_LV_LOG != 0
    lv_log_register_print_cb(my_print); /* register print function for debugging */
#endif

    tft.begin(); /* TFT init */
    tft.setRotation(3); /* Landscape orientation */

    tft.setTouch(calData);
    screenWidth  = tft.width();
    screenHeight = tft.height();
  
    lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);

    /*Initialize the display*/
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = 320;
    disp_drv.ver_res = 240;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.buffer = &disp_buf;
    lv_disp_drv_register(&disp_drv);

    lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);             /*Descriptor of a input device driver*/
    indev_drv.type = LV_INDEV_TYPE_POINTER;    /*Touch pad is a pointer-like device*/
    indev_drv.read_cb = my_touchpad_read;      /*Set your driver function*/
    lv_indev_drv_register(&indev_drv);         /*Finally register the driver*/

    /*Initialize the graphics library's tick*/
    tick.attach_ms(LVGL_TICK_PERIOD, lv_tick_handler);
    
    xSemaphore = xSemaphoreCreateMutex();

    snConverter();
    
    lv_main();
 
}

void particles(){
  
   sensirion_i2c_init();
    
    while (sps30_probe() != 0) {
      Serial.print(" Fallo del sensor de particulas\n");
      delay(500);
    }

      #ifndef PLOTTER_FORMAT
        Serial.print(" Particulas iniciado\n");
      #endif /* PLOTTER_FORMAT */
      
        ret = sps30_set_fan_auto_cleaning_interval_days(auto_clean_days);
        if (ret) {
          Serial.print("Error estableciendo el intervalo de días del autolimpiado\n");
          Serial.println(ret);
        }
      
        ret = sps30_start_measurement();
        if (ret < 0) {
          Serial.print("error starting measurement\n");
        }
        
        #ifdef SPS30_LIMITED_I2C_BUFFER_SIZE
          Serial.print(" Limitación del buffer del procesador");
        
          Serial.print("\n");
          delay(2000);
        #endif
}`

Results: 

```
SPS sensor probing successful
measurements started
PM  1.0: 4437.62
PM  2.5: 4692.62
PM  4.0: 4692.62
PM 10.0: 4692.62
NC  0.5: 30175.18
NC  1.0: 35249.48
NC  2.5: 35415.97
NC  4.0: 35428.58
NC 10.0: 35435.18
Typical partical size: 0.28

co2(ppm):953 temp(C):22.1 humidity(%):42.8
PM  1.0: 4430.68
PM  2.5: 4685.28
PM  4.0: 4685.28
PM 10.0: 4685.28
NC  0.5: 30127.98
NC  1.0: 35194.35
NC  2.5: 35360.58
NC  4.0: 35373.17
NC 10.0: 35379.75
Typical partical size: 0.28

co2(ppm):954 temp(C):22.1 humidity(%):42.9
PM  1.0: 4461.71
PM  2.5: 4718.09
PM  4.0: 4718.09
PM 10.0: 4718.09
NC  0.5: 30338.96
NC  1.0: 35440.81
NC  2.5: 35608.21
NC  4.0: 35620.88
NC 10.0: 35627.52
Typical partical size: 0.28
```

mourazo, 2021-07-28:
I closed the issue by mistake

winkj, 2021-07-28:
Hi mourazo,

Values in this order of magnitude point towards a 3.3V supply voltage. Can you double check the SPS30 is powered with 5V?

mourazo, 2021-07-29:
That was my mistake, thanks

closed 2021-07-29 by mourazo