Sensirion/arduino-avr-legacy-i2c-sps30 #36: Compatibility issue with Arduino Uno Wifi Rev 2

robertamancini90, 2022-03-29:
Hello, we are trying to run the libray with our Arduino Uno Wifi Rev 2. 
Unfortuntaley it does not work, and we have an error saying "Error compiling for board Arduino Uno WiFi Rev2."

It must be a board related error, because we have already tested the library with an Arduino UNO Rev 3 e it works perfectly.

Any idea on how to fix it? Thanks :)

winkj, 2022-03-29:
Could you enable verbose mode, and share the error message? https://support.arduino.cc/hc/en-us/articles/4407705216274-Use-verbose-output-in-the-Arduino-IDE

robertamancini90, 2022-03-29:
Hi @winkj ,
thanks for the prompt reply :)

I also attach here the picture that shows which board I have selected.

Here is the error message:
`Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\roberta.mancini\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\roberta.mancini\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\roberta.mancini\Documents\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=on -vid-pid=03EB_2145 -ide-version=10819 -build-path C:\Users\ROBERT~1.MAN\AppData\Local\Temp\arduino_build_973786 -warnings=none -build-cache C:\Users\ROBERT~1.MAN\AppData\Local\Temp\arduino_cache_800983 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\roberta.mancini\Documents\Arduino\libraries\arduino-sps-master\examples\sps30\sps30.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\roberta.mancini\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.Ard
[image]
uinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\roberta.mancini\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\roberta.mancini\Documents\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=on -vid-pid=03EB_2145 -ide-version=10819 -build-path C:\Users\ROBERT~1.MAN\AppData\Local\Temp\arduino_build_973786 -warnings=none -build-cache C:\Users\ROBERT~1.MAN\AppData\Local\Temp\arduino_cache_800983 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\roberta.mancini\Documents\Arduino\libraries\arduino-sps-master\examples\sps30\sps30.ino

Using board 'uno2018' from platform in folder: C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\hardware\megaavr\1.8.7

Using core 'arduino' from platform in folder: C:\Users\roberta.mancini\Documents\ArduinoData\packages\arduino\hardware\megaavr\1.8.7

Detecting libraries used...

"C:\\Users\\roberta.mancini\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 "-IC:\\Users\\roberta.mancini\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino/api/deprecated" "-IC:\\Users\\roberta.mancini\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino" "-IC:\\Users\\roberta.mancini\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.7\\variants\\uno2018" "C:\\Users\\ROBERT~1.MAN\\AppData\\Local\\Temp\\arduino_build_973786\\sketch\\sps30.ino.cpp" -o nul

Alternatives for sps30.h: [sensirion-sps@0.9.0 arduino-sps-master@1.0.0]

ResolveLibrary(sps30.h)

  -> candidates: [sensirion-sps@0.9.0 arduino-sps-master@1.0.0]

[... cut ...]
C:\Users\roberta.mancini\Documents\Arduino\libraries\arduino-sps-master\i2c_master_lib.cpp:712:26: note: suggested alternative: 'TWI0'

Multiple libraries were found for "sps30.h"

 Used: C:\Users\roberta.mancini\Documents\Arduino\libraries\arduino-sps-master

 Not used: C:\Users\roberta.mancini\Documents\Arduino\libraries\sensirion-sps

Using library arduino-sps-master at version 1.0.0 in folder: C:\Users\roberta.mancini\Documents\Arduino\libraries\arduino-sps-master 

exit status 1

Error compiling for board Arduino Uno WiFi Rev2.

`

winkj, 2022-03-30:
thanks for sharing that.

I believe I see what's happening: we added an internal I2C library to work around an I2C buffer size limit in the Wire library in AVR Arduinos (like the regular Uno). It appears that the Uno Wifi is setting the ``__AVR__`` define, which we use to enable said workaround (happening here: https://github.com/Sensirion/arduino-sps/blob/master/sensirion_arch_config.h#L39). 

The good news is that from what I can see, the workaround is not needed for MegaAVR boards, as they already support an I2C buffer of 128 bytes: https://github.com/arduino/ArduinoCore-megaavr/blob/master/libraries/Wire/src/Wire.h#L28

The bad news is that to fix this properly, we'll have to do a bunch of testing to see which AVR architectures need this.

For the time being, I've created a branch with a fix for this here: https://github.com/Sensirion/arduino-sps/tree/winkj-temp-fix-atmega4809. The change is very local, so it should be easy to do this manually in your Arduino library folder; the diff can be found here: https://github.com/Sensirion/arduino-sps/commit/cf93dc5eec876f6c60ca31bc7be0c20967c74f3a

Unfortunately, I don't have a Uno Wifi Rev 2 board, so while the fix makes this build for this board, I could not test whether communication works properly. If you could apply the change above and let me know whether that works that would be highly appreciated.

Also, since you mentioned also using a Uno (Non-wifi) Rev 3, this patch should continue to work on that platform, so that you can safely modify your ``Arduino/libraries/sensirion-sps/sensirion_arch_config.h`` without risking breaking other platforms.

Let me know if you have any questions!
Best, Johannes

winkj, 2022-03-30:
awesome, thanks for the report and testing the workaround. I will close this issue, since I opened a separate issue (#37) for us to keep track of the change

closed 2022-03-30 by winkj

matteodanelli, 2022-04-21:
Sorry to comment on this issue. I'mve experienced the same issue and solved with the branch linked.
However, the sensor fails (sps30_probe() return zero).

Do you know if the code example code should change too or pin connection has some differences with, for example, Arduino uno?

winkj, 2022-04-21:
`sps30_probe()` returns an error code other than zero if it fails. If it returns zero, the call was successful (zero is the equivalent of `NO_ERROR`, https://github.com/Sensirion/arduino-sps/blob/master/sensirion_common.h#L41).

matteodanelli, 2022-04-21:
@winkj right, bad typo. It returns some sort of errors (value different from zero)

matteodanelli, 2022-04-21:
@robertamancini90  Did you had the same issue?

winkj, 2022-04-21:
Roberta reported earlier that the change worked "perfectly", so I doubt that they hit that same issue.

So regarding your second question, no changes should be necessary to the example when using the branch, since the change is purely internal. As I mentioned before, I unfortunately don't have access to a MegaAvr, so can't really test myself.

To debug communication issues, I would recommend starting with the I2C scanner (https://playground.arduino.cc/Main/I2cScanner/) to make sure the connection is correct. If the I2C scanner doesn't find the sensor, it's likely a wiring/connection issue. 

If the I2C scanner detects the sensor but sps30_probe() fails to initialize it properly, please open a new issue. Thanks!

matteodanelli, 2022-04-21:
Ok thanks! I2C detects the sensor at address 0x60, so it's fine.
Therefore I'll open a new issue.

winkj, 2022-04-21:
@matteodanelli 0x60 is likely something else on the bus, the SPS30 would answer on 0x69

matteodanelli, 2022-04-21:
So, something wrong with wiring? 
I follow this scheme:
Pin1: 5v
Pin2: A4
Pin3: A5
Pin4: GND
Pin5: GND

winkj, 2022-04-21:
Based on https://content.arduino.cc/assets/Pinout-UNOwifirev2_latest.png, SDA/SCL would be pins D20/D21, not A4/A5

matteodanelli, 2022-04-21:
Oh, you saved me. I missed this difference with the standard Uno. Now it works. Thanks a lot!

winkj, 2022-04-21:
@matteodanelli awesome, glad to hear the SPS30 is communicating, and thanks for letting us know!