Sensirion/arduino-avr-legacy-i2c-sps30 #39: Compiler warning on ESP32

kaysond, 2023-02-28:
I'm getting the following warning when compiling:
```
In file included from c:\Users\aram\Documents\Arduino\libraries\sensirion-sps\sensirion_hw_i2c_implementation.cpp:78:
C:\Users\aram\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\Wire\src/Wire.h: In function 'int8_t sensirion_i2c_read(uint8_t, uint8_t*, uint16_t)':
C:\Users\aram\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\Wire\src/Wire.h:127:13: note: candidate 1: 'uint8_t TwoWire::requestFrom(int, int)'
     uint8_t requestFrom(int address, int size);
             ^~~~~~~~~~~
C:\Users\aram\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\Wire\src/Wire.h:125:13: note: candidate 2: 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t)'
     uint8_t requestFrom(uint8_t address, uint8_t size);
             ^~~~~~~~~~~
```

The problem is that `sensirion_i2c_read()`'s `count` argument is `uint16_t` but none of the `Wire.h` `requestFrom()` overloads support such a large size read request. Do you really expect >256 bytes? The easiest solution would just be to change the argument to `uint8_t`, which is what's happening anyways.

qfisch, 2023-02-28:
Your reasoning seems correct to me.
I will bring up this issue internally.
Thanks for bringing that up, I will keep you up to date

winkj, 2023-02-28:
This specific interface is coming from our embedded-common repository (https://github.com/Sensirion/embedded-common/blob/1ac7c72c895d230c6f1375865f3b7161ce6b665a/sensirion_i2c.h#L74). I'm happy to change this for the SPS30 separately, but it may make sense to update embedded-common as well

winkj, 2023-02-28:
Update: a fix for this was submitted previously with PR #21, and we requested this to be applied to embedded-common instead; from what I can see, this has not happened

@qfisch  @psachs would you prefer a local fix, or to change this in embedded-common?

qfisch, 2023-03-01:
@winkj , I think updating embedded-common makes more sense.
I will create an issue internally and leave this one open so we don't indefinitely push it back.

@kaysond Thanks for bringing that up to us again. This time we hopefully come to a fix.

kaysond, 2023-04-17:
@qfisch - any updates? I'm happy to open a PR here or embedded-common

psachs, 2023-04-19:
The pull request got merged and the issue should be resolved. Let us know if have additional questions.

closed 2023-04-19 by psachs

psachs, 2023-04-20:
sure, consider it done