Sensirion/arduino-avr-legacy-i2c-sps30 #41: Differt values Plotter Format and Limited_I2C_Buffer_Size

barnycloud, 2023-04-08:
I am using the SPS30.ino-sketch on a esp8266. The SPS30-Device is connected via I2C to the microcontroller.
Everything works fine, the sensor probing is successful  but I get the outputs:
```
PM  1.0: 3.05
PM  2.5: 3.23
PM  4.0: 3.23
PM 10.0: 3.23
NC  0.5: 20.98
NC  1.0: 24.29
NC  2.5: 24.37
NC  4.0: 24.38
NC 10.0: 24.39
Typical partical size: 0.37
```
What are the differences between values denoted with 'NC' and 'PM'? The values differ almost by a factor of 10.  Are these values given in 'ppm'?
Any help appreciated.

barnycloud, 2023-04-09:
As far as I can see from the Sensirion data sheet (of the SPOS30), MC means mass concentration (resp. mass concentration size range) and NC number concentration (resp. nember concentration size range). It would be helpful, at least for me, if the structure of the sketch code would contain a bit more comments.

sdmueller, 2023-04-19:
Hi @barnycloud

Your assumptions are correct.
Thank you for the input. We will consider adding appropriate comments to the sketch code.

Let us know if you have further questions.

closed 2023-04-19 by sdmueller