Sensirion/arduino-avr-legacy-i2c-sps30 #48: Strange readings

NetForces, 2023-12-21:
I'm using the SPS-30 connected to an M5Stack CoreInk. 

I had strange reading in my code so decided to try the example in this library and I'm getting the same strange readings:

```
PM  1.0: 15509.48
PM  2.5: 57628.93
PM  4.0: 91669.92
PM 10.0: 108974.59
NC  0.5: 0.00
NC  1.0: 75761.80
NC  2.5: 117888.16
NC  4.0: 123259.44
NC 10.0: 124015.24
Typical particle size: 1.46

PM  1.0: 15510.95
PM  2.5: 57571.49
PM  4.0: 91563.79
PM 10.0: 108843.69
NC  0.5: 0.00
NC  1.0: 75841.88
NC  2.5: 117908.30
NC  4.0: 123271.95
NC 10.0: 124026.68
Typical particle size: 1.45

PM  1.0: 15502.96
PM  2.5: 57495.44
PM  4.0: 91431.91
PM 10.0: 108683.45
NC  0.5: 0.00
NC  1.0: 75856.61
NC  2.5: 117854.21
NC  4.0: 123209.05
NC 10.0: 123962.59
Typical particle size: 1.46
```

The only difference in the example is that I added a `Wire.begin(25,26);` because these are the pins I am using for the I2C bus.

So the communication works fine, but the values don't make any sense...

winkj, 2023-12-21:
Hi there, we've seen similar issues when the sensor is powered at 3.3V... could you double check the supply voltage for the sensor?

NetForces, 2023-12-21:
Indeed, it's supplied with 3.3v. Let me change that a bit later this afternoon to see if I get more "normal" results.

NetForces, 2023-12-21:
Yeah much better !

```
PM  1.0: 0.37
PM  2.5: 1.26
PM  4.0: 1.98
PM 10.0: 2.35
NC  0.5: 0.23
NC  1.0: 1.92
NC  2.5: 2.81
NC  4.0: 2.92
NC 10.0: 2.94
Typical particle size: 1.63

PM  1.0: 0.34
PM  2.5: 1.17
PM  4.0: 1.84
PM 10.0: 2.18
NC  0.5: 0.17
NC  1.0: 1.74
NC  2.5: 2.57
NC  4.0: 2.67
NC 10.0: 2.69
Typical particle size: 1.63
```

Thanks for the quick reply. Closing.

closed 2023-12-21 by NetForces